mruettgers/SMLReader #1: EspSoftwareSerial change after 5.0.4

tunip, 2019-12-10:
Hi,

since version 5.0.4 of EspSoftwareSerial something changed and compile of SMLReader will fail.

See issue:
https://github.com/plerup/espsoftwareserial/issues/123

Would you adjust the chnages in EspSoftwareSerial to your project? Or should we stay at EspSoftwareSerial@5.0.4 in platformio.ini?

mruettgers, 2019-12-10:
Hey,

thank you for the feedback.
I'll have a look on it later on, but I don't see a reason for not adjusting the code.
But Pull-Requests are also welcome ;-).

Cheers,
Michael

mruettgers, 2019-12-10:
@tunip I just pushed a new version that should fix the issue. 
Could you please give this a try?

Thanks.

closed 2019-12-10 by mruettgers

tunip, 2019-12-11:
@mruettgers compiling and flashing my wemos mini d1 works fine. Need to wait for the BPW40 (this week) then I can give you detailed feedback about the function.

Many thanks

mruettgers, 2019-12-11:
Ok, fine.
Did you already request the PIN from your provider to unlock the SML interface of your meter using a flash light?

tunip, 2019-12-11:
I am using vzlogger which works fine right now. But I would like to move this task to a microcontroller. And your project fits this need perfectly.

tunip, 2019-12-15:
First test looks good.

I only missing "1.8.0" power_in. It sends only "-1".
power_current is fine.

config.h:
```
static const metric METRICS[] = {
    {"power_in", {0x77, 0x07, 0x01, 0x00, 0x01, 0x08, 0x00, 0xFF}},
    {"power_current", {0x77, 0x07, 0x01, 0x00, 0x0F, 0x07, 0x00, 0xFF}}};
```

MQTT Output
```
-1
599600
-1
600900
-1
587000
-1
589300
-1
586500
-1
586400
```

mruettgers, 2019-12-15:
> First test looks good.
> 
> I only missing "1.8.0" power_in. It sends only "-1".
> power_current is fine.
> 
> config.h:
> 
> ```
> static const metric METRICS[] = {
>     {"power_in", {0x77, 0x07, 0x01, 0x00, 0x01, 0x08, 0x00, 0xFF}},
>     {"power_current", {0x77, 0x07, 0x01, 0x00, 0x0F, 0x07, 0x00, 0xFF}}};
> ```

What meter model do you have?

mruettgers, 2019-12-15:
If you have a serial monitor attached, please enable verbose debugging in `platform.io` (`-DSERIAL_DEBUG_VERBOSE=true`) and paste the SML response data from the meter.

tunip, 2019-12-16:
The meter is a ED300L.

Debug output:
```
State is 'wait_for_start_sequence'.
Start sequence found.
State is 'read_message'.
End sequence found.
State is 'read_checksum'.
Message has been read.
----DATA----
1B 1B 1B 1B 01 01 01 01 76 07 00 1E 03 F8 7D 98 
62 00 62 00 72 63 01 01 76 01 01 07 00 1E 0F A3
D4 88 09 04 03 9F C8 DE 5D 25 9E 01 01 63 4E B0
00 76 07 00 1E 03 F8 7D 99 62 00 62 00 72 63 07
01 77 01 09 04 03 9F C8 DE 5D 25 9E 01 72 62 01
65 0F A3 E6 BB 77 77 07 81 81 C7 82 03 FF 01 01
01 01 04 45 4D 48 01 77 07 01 00 00 00 09 FF 01
01 01 01 09 04 03 9F C8 DE 5D 25 9E 01 77 07 01
00 01 08 00 FF 63 02 82 01 62 1E 52 FF 56 00 15
27 CD A8 01 77 07 01 00 01 08 01 FF 01 01 62 1E
52 FF 56 00 00 00 00 00 01 77 07 01 00 01 08 02
FF 01 01 62 1E 52 FF 56 00 15 27 CD A8 01 77 07
01 00 0F 07 00 FF 01 01 62 1B 52 FF 55 00 00 15 
5D 01 77 07 81 81 C7 82 05 FF 01 01 01 01 83 02
5C 97 D7 54 25 1B 00 E7 4C AE 7C A0 30 EE B5 D5
68 1B 16 46 99 2D 8D FF 1D 63 DE 53 61 37 D1 CE 
41 F7 EC 49 14 A2 F4 69 C7 FA 9C 8F B4 E7 BB 1C
01 01 01 63 87 99 00 76 07 00 1E 03 F8 7D 9C 62
00 62 00 72 63 02 01 71 01 63 BE 53 00 00 00 00
1B 1B 1B 1B 1A 03 60 B5
---END OF DATA---
State is 'process_message'.
Found metric 'power_in.'
Found metric 'power_current.'
Publishing message to 'smartmeter/metric/power_in/value':
-1
Published metric 'power_in':
  Value: 354930088
  Unit: 30
  Scaler: -1
Publishing message to 'smartmeter/metric/power_current/value':
546900
Published metric 'power_current':
  Value: 5469
  Unit: 27
  Scaler: -1
State is 'wait_for_start_sequence'.
Start sequence found.
State is 'read_message'.
End sequence found.
```

`Value: 354930088` is correct, but not published.

Many Thanks

mruettgers, 2019-12-16:
I just pushed an update, please give it a try.
The size of the datatype used for the value in the MQTT client had been too small.

tunip, 2019-12-16:
Now it works fine. Thanks for your great and fast help.

commit 8341617a (2019-12-10): Fixes #1 among some other improvements
commit a4bc6027 (2019-12-10): Working on a fix for #1

Fixes #1 among some other improvements