mruettgers/SMLReader #2: Hardware question, USB-IR-Head

Hoelli4C, 2020-01-31:
Hello,

I previously had a Raspbi with the volkszaehler.org solution installed. I used an USB-IR-Head (https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-usb-ausgang) for this.

I would like to use your solution on a Wemos D1 Mini. 

My question: Do you think it would be possible that I just create a small adapter to connect 5V and Data and then use my old IR-Head OR would you recomment to just buy a BPW 40 and build a new one?

Thanks for your help.

mruettgers, 2020-01-31:
Hey,

connecting the USB-IR-Head won't work because it is based on a USB-to-UART bridge that does not provide direct access to the underlying phototransistor.

Of course you could disassemble the read head and use the built-in phototransistor, but given the low price I would buy a BPW 40.

tunip, 2020-01-31:
BPW 40 with Wemos D1 Mini runs fine in my environment. I also used a USB-IR-Head from volkszaehler.org before.

Sometimes (~2 per week) I have some extreme spikes from the power_current reads, but i filter this in Home Assistant (value could not be negative or higher then e.g. 20000 watt).

mruettgers, 2020-01-31:
@tunip 
Hmm, that sounds weird.
I could not detect any spikes in my records so far.
If the problem persists, can you please create a new issue so that we can try to figure out what's going on there?

closed 2020-04-09 by mruettgers

mruettgers, 2020-04-14:
@tunip If you are still encountering spikes you may want to test the current version. Instead of using my own implementation for decoding the SML data  I switched over to use libSML from Volkszähler. That should give us more reliable results.
Beside this it is now possible to attach multiple reading heads.

tunip, 2020-04-14:
@mruettgers thx for the feedback. Yes I will try the new libSML version later this week and let you know the result.