mruettgers/SMLReader #3: Static Sensor Readings

ostpol, 2020-04-02:
Michael,

thanks for building SMLReader and making it available to the public!

I've just set up my SMLReader with a selfmade IR head that was used with Volkszähler and a Raspberry Pi before. The meter is a DWS 7412.1.

Just to be clear:

`power_current` is the currently consumed power?
`power_out` is the overall consumption?
`power_in` is produced power?

SMLReader receives data and the connection to my MQTT server works, but the readings MQTT receives are static:

`iot/smartmeter/metric/power_current/value` is allways 27000
`iot/smartmeter/metric/power_out/value` is allways 9223372036854775807  (should be arround 5460)

I noticed, that `power_out` readings are going from 5460 to 54602362 to 546023624746 to 9223372036854775807 after boot.

Serial output is:

```
State is 'wait_for_start_sequence'.
Start sequence found.
State is 'read_message'.
End sequence found.
State is 'read_checksum'.
Message has been read.
----DATA----
1B 1B 1B 1B 01 01 01 01 76 05 48 32 66 04 62 00 
62 00 72 63 01 01 76 01 01 02 31 0B 0A 01 44 5A 
47 00 02 81 87 15 72 62 01 65 03 E7 DD A9 62 02 
63 76 31 00 76 05 49 32 66 04 62 00 62 00 72 63 
07 01 77 01 0B 0A 01 44 5A 47 00 02 81 87 15 07 
01 00 62 0A FF FF 72 62 01 65 03 E7 DD A9 74 77 
07 01 00 60 32 01 01 01 72 62 01 62 00 62 00 52 
00 04 44 5A 47 01 77 07 01 00 60 01 00 FF 01 72 
62 01 62 00 62 00 52 00 0B 0A 01 44 5A 47 00 02 
81 87 15 01 77 07 01 00 01 08 00 FF 64 1C 01 04 
72 62 01 62 00 62 1E 52 FF 65 03 6F A4 71 01 77 
07 01 00 10 07 00 FF 01 72 62 01 62 00 62 1B 52 
FE 53 99 36 01 01 01 63 0B 38 00 76 05 4A 32 66 
04 62 00 62 00 72 63 02 01 71 01 63 A1 88 00 00 
1B 1B 1B 1B 1A 01 4D F3 
---END OF DATA---
State is 'process_message'.
Found metric 'power_in.'
Found metric 'power_current.'
Publishing message to 'iot/smartmeter/metric/power_in/value':
30000
Published metric 'power_in':
  Value: 30
  Unit: 98
  Scaler: 0
Publishing message to 'iot/smartmeter/metric/power_out/value':
0
Published metric 'power_out':
  Value: 5460236247467294720
  Unit: 30
  Scaler: -30
Publishing message to 'iot/smartmeter/metric/power_current/value':
27000
Published metric 'power_current':
  Value: 27
  Unit: 98
  Scaler: 0
```

The readings in the serial output are static too.

Is the problem related to the `scaler`? Why are `power_out` and especially `power_current` static?

I'm a mess in reading (or writing) code. That's why I'm asking here instead of just read the code by myself.

Best reagrds und bleib Gesund,
T

mruettgers, 2020-04-02:
Hey T,

thank you for your detailed report.

Regarding to your question

> Just to be clear:
> 
> power_current is the currently consumed power?
> power_out is the overall consumption?
> power_in is produced power?

Well, it depends on the metrics defined in `config.h`.
The name can be choosen freely and is just used as part of the MQTT topic.

The manual of your meter says, that it provides the overall consumption by using the OBIS identifier 1.8.0, which should result in a config like:

// DGZ  DWS74/DVS74
static const metric METRICS[] = {
    {"consumption", {0x77, 0x07, 0x01, 0x00, 0x01, 0x08, 0x00, 0xFF}}};

Related to the debug output, you seem to be using the default config without any adjustments, and thus SMLReader is expecting metrics not beeing provided by the meter.

I did not really take care of this case which is now resulting in publishing an uninitialized variable exponentiated by another uninitialized variable for the metric "power_out".
So this is indeed a bug :-).

But nevertheless I'm wondering why the value of "power_in" is not being updated properly, although it refers to the value of the OBIS identifier 1.8.0 (assuming you are using the config of my meter).

I'll have a deeper look at the binary data as soon as possible to try to figure out what's going on here.

Maybe it is related to some known issues with those kind of meters:
* https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/dzg_dvs74
* https://github.com/jmberg/libsml/commit/81c4026e3d94f7a384cdd89f62a727b83269cdec

Danke, bleib Du auch gesund.

Many regards,

Michael

ostpol, 2020-04-03:
Hi Michael,

thank you for your quick response.

Yes, config and code are 100% untouched. I thought it was worth a try.

Let me know if you need more debug output or other informations. I'll try to dig into the specifics of my meter too.

mruettgers, 2020-04-07:
Hey Tony,

the good news is, that the SML-Datagram seems to contain the correct value for your consumption (57648241).
The bad news is that I did not have the time to figure out what's going on yet, but I'll keep you posted.

I'm also taking into consideration to get rid of the static configuration in favor of parsing the whole SML message to publish all information available.
But this will take some time, i guess.

Michael

mruettgers, 2020-04-07:
My tests with https://github.com/volkszaehler/libsml and your test data look promising so far..

```bash
srv-01 ➜  /var/tmp  docker run -it --device /dev/ttyUSB0 -v /var/tmp:/src --rm mruettgers/esptool ash -c "esptool --port /dev/ttyUSB0 write_flash -fm dout 0x00000 /src/smltest1.bin && miniterm.py /dev/ttyUSB0 115200"
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:0d:8e:ab:af:4e
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0340
Compressed 273328 bytes to 197206...
Wrote 273328 bytes (197206 compressed) at 0x00000000 in 17.6 seconds (effective 124.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Started...
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 01 44 5a 47 00 02 81 87 15 #

1-0:1.8.0*255#5764824.1#Wh
1-0:16.7.0*255#392.22#W
```

mruettgers, 2020-04-08:
@ostpol I just pushed a first working draft  to https://github.com/mruettgers/SMLReader/tree/feature/refactoring
Feel free to test.

```
MB-Monty ➜  ~  mosquitto_sub -h 10.4.32.103 -v -t smartmeter/mains/#
smartmeter/mains/sensor/1/obis/1-0:1.8.0*255/value 3468847.3
smartmeter/mains/sensor/1/obis/1-0:2.8.0*255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.1*255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:2.8.1*255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.2*255/value 3468847.3
smartmeter/mains/sensor/1/obis/1-0:2.8.2*255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:16.7.0*255/value 298.6
```

ostpol, 2020-04-08:
@mruettgers Wow! I'll test it tomorrow.

ostpol, 2020-04-09:
@mruettgers I just flashed the new version. Works like a charm.

Vielen Dank, lass dir den Kaffee schmecken & bis bald!

closed 2020-04-09 by ostpol