mruettgers/SMLReader #4: [FR] input for "old" pulse based sensors / meters

hamster65, 2020-07-26:
Hi Michael,

thanks for publishing this project! We will get an electronic power meter installed during the next weeks, and I already have everything prepared, including a 3d printed IR head etc.

Currently, I use an Atmega128 controller to receive impulses from our gas meter (reed sensor), water meter (reflex light sensor) and ferraris energy meter (green led+photo transistor). Since the water and gas meter will not be changed, it would be nice to connect those sensors to the Wemos module to get an output by mqtt. 

Unfortunately, there is no periodic output from those meters when there is no or very low consumption. Right now, I use a fixed interval and I output how many pulses have been detected during that time but this is not good for live monitoring.

I can think of 3 ways to do this:
- Fixed time (how long?)
- MQTT message  including time in ms since last pulse on every pulse (both rising and falling edge, because that could double the resolution, depending on the meter/ sensor). This would allow to count pulses (and therefore exact total consumption) and it would also allow to get a very high resolution and fast updates. Rate limiting would be important to prevent defective sensors from generating too many mqtt messages. The home automation server would have to calculate consumption depending on pulses and / or timestamps and deal with the situation when no more pulse are received, so that complexity would not be an issue of  the firmware).
- Just send an mqtt message on rising and falling edge (with rate limiting). Most simple approach, but without time reference, not accurate for live monitoring.

I am looking forward to reading your thoughts on this.

Regards,
hamster 65

mruettgers, 2020-08-02:
Hi hamster 65,

I'm sorry I didn't get back to you sooner. My two children and the many tasks at the house take too much of my free time. 
Therefore, I will unfortunately not be able to implement this FR in the near future.
I also lack the appropriate hardware to be able to test the whole thing meaningfully.

I did already think about making my analog water meter smart, but I would try to solve the problem with computer vision like described in this project:
https://www.kompf.de/cplus/emeocv.html

Regards,

Michael

hamster65, 2020-08-03:
Thanks Michael!

closed 2020-08-03 by hamster65

mruettgers, 2020-08-04:
@hamster65 I just took notice of a project for old meters on Instagram: https://github.com/Eisbaeeer/Ferraris_MQTT_Energy_Counter_Meter_TCRT5000
It might be worth a look.

Cheers, 
Michael