mruettgers/SMLReader #5: The meaning of MQTT topics

HakamSaffour, 2020-09-22:
Hi,
Can you explain what is the meaning of the MQTT topics

smartmeter/mains/sensor/1/obis/1-0:1.8.0/255/value 3546245.9
smartmeter/mains/sensor/1/obis/1-0:2.8.0/255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.1/255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:2.8.1/255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.2/255/value 3546245.9
smartmeter/mains/sensor/1/obis/1-0:2.8.2/255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:16.7.0/255/value 451.2

on my meter, I am getting the following values
[image]

Thanks in advance

ostpol, 2020-09-22:
Hi @HakamSaffour,

smartmeter/mains/sensor/1/obis/**1-0:1.8.0**/255/value 3546245.9

The bold part ist the OBIS identifier. 1.8.0 should be the overall consumption of your meter.

Other identifiers are documented in this (german) [document](https://www.bundesnetzagentur.de/DE/Service-Funktionen/Beschlusskammern/BK06/BK6_81_GPKE_GeLi/Mitteilung_nr_62/Anlagen/Codeliste_OBIS_Kennzahlen_2.2g.pdf?__blob=publicationFile&v=2). 

I hope it helps a bit.

– T

mruettgers, 2020-09-22:
Hi @HakamSaffour,

in addition to @ostpol's answer:

1-0:16.7.0 should be the current consumption (Wirkleistung Momentan).

Michael

HakamSaffour, 2020-09-23:
Vielen Dank @ostpol  und @mruettgers

closed 2020-09-23 by HakamSaffour