mruettgers/SMLReader #6: Improvement suggestion: Change of MQTT topic

HakamSaffour, 2020-09-23:
Scenario 1: After setting up everything and getting some results, I wanted to change the MQTT topic, I entered the local IP address of the SMLReader, then I changed the MQTT topic , then I clicked on 'Save'

Result: the SMLReader can not connect to the MQTT broker

Scenario 2: did the same as in scenario 1 and I also retyped the MQTT password, then clicked 'Save'
Result: the SMLReader do connect to the MQTT broker

[image]

Potential improvement: make sure that MQTT password is not lost by changing the configuration

mruettgers, 2020-09-24:
Hey @HakamSaffour,

thank you for your feedback.
I'll have a look at as soon as possible, but I guess this is an issue with the third-party-library [IotWebConf](https://github.com/prampec/IotWebConf).

Michael

mruettgers, 2021-04-23:
I'm closing this issue for now, as I don't currently have time to debug the third-party library. I admit that the problem is annoying, but it can be fixed with a workaround and only occurs during the initial setup.

closed 2021-04-23 by mruettgers