mruettgers/SMLReader #9: Web-interface requires login

Syntoxr, 2020-10-08:
I think it isn't an issue but, I couldn't figure out what the login credentials for the web-interface are.

Thanks for your work!

mruettgers, 2020-10-08:
Hey @Syntoxr,

user `admin` and the initially set AP password should work.
At least they worked for me right now.

Michael

mruettgers, 2020-10-08:
I just updated the docs accordingly

Syntoxr, 2020-10-09:
Hi @mruettgers ,

Thanks for your quick reply.
I tried standard usernames and passwords before (user `admin` and password `AP password` too) but that didn't work.
I'll try again later that day.

mruettgers, 2020-10-09:
With `AP password` I meant the password you've configured during the initinal setup.

Syntoxr, 2020-10-09:
After flashing again it worked. I think I mistyped the password when setting it.
Thanks for your help.

closed 2020-10-09 by Syntoxr