mruettgers/SMLReader #10: No data receiving

sideshowbob100, 2020-10-31:
I only get the infomation that the SMLReader in online and connected but there a no more data points available.
I flashed the SMLReader_D1mini_v2.1.5.bin,
then I connected the ESP to my WiFi and setup the IP, User and Password of my MQTT in iobroker.
Do you have any idea what I did wrong?

[image]

mruettgers, 2020-10-31:
Hey,

the data points will be published via MQTT only in case SMLReader receives data from the meter.
The builtin LED of the D1 will flash 3 times every time the data is being read from the meter to help positioning the sensor right.

Michael

sideshowbob100, 2020-10-31:
Hi Michael,
thx for the fast answer. Ok on Monday they will install the smart meter.
I will check it then!

Sören

sideshowbob100, 2020-11-04:
Ist doesn't work but I red that my smart meter sends the code in OBIS standard. Does the SML Reader Software also support OBIS?

mruettgers, 2020-11-04:
What meter model do you have? Did you unlock it with the PIN code provided the provider? And yes, OBIS is the way SMLReader interprets the data.

sideshowbob100, 2020-11-04:
It is a Logarex LK13BEB03039.
Yes I unlocked it with a code and I also deactivated the PIN.
The IR LED of the smart meter is flashing, I check it with a handy cam.

mruettgers, 2020-11-04:
Hm. So I guess the LED of the D1 does not flash at all?
There can be several reasons causing this behaviour:
* Your meter does not speak SML but rather another protocol
* The baud rate or parity settings mismatch the ones of your meter
* The phototransistor is broken or the wiring is wrong

Do you have any chance to log the serial output of the meter using different serial settings?
If you have access to the serial console output of the D1 when attached to the meter it might be an easy task to write a data logger dumping out anything received from the meter at different baud rates to figure out what's being sent.

sideshowbob100, 2020-11-05:
I measured the voltage over the photo transistor and it changed from 3.3v to 0v when I changed it von dark to light so I guess it works fine.
At the moment I have no idea how I can get access to the serial console output but I will check it.
THX!

closed 2021-01-03 by mruettgers

abbe79, 2021-11-01:
Hi,

I got a Logarex LK13BE803049 installed a few days ago, here's a few lines of output I dumped with `screen /dev/tty.usbserial-00000000 9600` using an USB/serial converter.

@mruettgers, could you please have a look at it? I looks like the standard SML text protocol. But I already failed at trying to read the data using Tasmota. Can smlreader parse the text protocol or does it use the binary form?

@sideshowbob100, I had to unlock the meter by typing in the PIN that I could look up at the Bayernwerk Netz customer website. The meter also allows to send an extended number of values, which can be done by switching the menu item "info" to "on".

Here's a link to the German manual of the meter: https://www.stadtwerke-burgdorf-netz.de/_Resources/Persistent/9450d40cdc3d62d8de38a3e4b06ad5d6805c87b4/Gebrauchsanleitung_LK13BE8030x9.pdf

And here's the small dump I saved: [logarex.txt](https://github.com/mruettgers/SMLReader/files/7452854/logarex.txt)

Thanks!

abbe79, 2021-11-01:
Looks like smlreader does also not like the text version of the protocol:

```
Hello from 0053C8FF, running SMLReader version 2.2.0.

Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
```

abbe79, 2021-11-01:
Hah! If I read the data with the correct 7E1 (7 bit data, even parity bit, one stop bit) parameters, it gets better:
```
1-0:96.1.0*255(001LOG0064026774)
1-0:1.8.0*255(000016.1347*kWh)
1-0:1.8.1*255(000006.9047*kWh)
1-0:1.8.2*255(000009.2300*kWh)
1-0:2.8.0*255(000000.0000*kWh)
1-0:0.2.0*255(ver.03,432F,20170504)
1-0:96.90.2*255(61BF)
1-0:97.97.0*255(00000000)
!
/LOG5LK13BE803049
```