mruettgers/SMLReader #11: [Feature Request] Add interval function

tunip, 2020-11-26:
Would like to ask for a feature request to add a configurable value to set an interval in seconds for updating the mqtt topics.
My meter updates its values every 1-2 seconds. But for my use case, every 15 seonds is more then enough.

Thx

emmrichd, 2020-12-10:
For the time being, you could add a delay in the loop, couldn't you? Like delay(15000);

tunip, 2020-12-10:
@emmrichd In main.cpp I tried at several places in "void loop" but does not work.
As a workaround I added a delay in MqttPublisher.h, which works for me.

```
    {
      // Something failed
      DEBUG("Connection to MQTT broker failed.");
      DEBUG("Unable to publish a message to '%s'.", topic);
      return;
    }
    DEBUG("Publishing message to '%s':", topic);
    DEBUG("%s\n", payload);
    client.publish(topic, payload);
    delay(1500);
  }
};

#endif
```

mruettgers, 2020-12-10:
I did not have the time to test it by myself yet, but feel free to do so ;-)

https://github.com/mruettgers/SMLReader/tree/feature/interval
The interval can be set in `config.h`

emmrichd, 2020-12-11:
Thanks a lot for the quick fix, Michael!
However, tunip's way is working, your seems not to. I tried different intervals.

mruettgers, 2020-12-11:
Well, that's strange.
I just did a test with an interval of 0 and another with 15 and it worked as expected.

emmrichd, 2020-12-11:
Ok, I will try again.

mruettgers, 2020-12-11:
Please note that the flashing of the LED is not an indicator.
The led will continue flashing as usual.
Only the MQTT messages are being throttled.

emmrichd, 2020-12-11:
Ok, sorry, I confirm, it works like a charm! Thanks a lot!

closed 2021-01-28 by mruettgers