mruettgers/SMLReader #12: Using pin RX

emmrichd, 2020-12-10:
Hello,

thank you for sharing this project!
I wanted to use the RX pin, but I don't get any data. The head is working for sure, since I used it with esp-link. Also, the LED flashes. On mqtt-side, I get the "Hello" message, but no data from my EMH meter.
Does this pin not work for your software? Have I chosen the wrong pin name?
Compilation tells me success.
Regards, Dieter

mruettgers, 2020-12-10:
Hey Dieter,

the TX&RX pins are used for being attached to a serial console for debugging output.
Internally they might also be used for flashing, but I'm not sure about that.

Do you have another GPIO pin available for testing?
What type of ESP8266 device are you using?

Michael

emmrichd, 2020-12-10:
Hello,

thanks for the quick reply. It is a Wemos D1 mini.
No problem, I move to D2, I just have to resolder it. Thanks.
Dieter

emmrichd, 2020-12-10:
Ok, fixed it, works now, thanks.

closed 2020-12-10 by mruettgers