mruettgers/SMLReader #16: Waiting for Sequence

realfire23, 2021-04-29:
Not getting any further than Hello From XXX running SMlReader.

In the debug it is waiting for Sequence. Tried a couple of BPW 40 :(

mruettgers, 2021-04-29:
What's the model of the meter you are trying to read?

realfire23, 2021-04-29:
hi,

thank you for the fast reply.

Its a Landis+Gyr E220.
 

One question is there any way to check if the trasnsistor is working before I attach it to the smart meter ?

mruettgers, 2021-04-29:
Did you unlock your meter by entering the PIN code provided by the provider via a flashlight?

I'm working on a new version of SMLReader that will allow debugging (including outputting RAW data read by the photostransitor) via web interface.
And I'm also planning to build a small application that will make a ESP8266 emulating a meter by providing SML data via an LED.

realfire23, 2021-04-29:
The pin has been entered.

Iam pretty sure Iam messing something up :D for debugging possibilities are always welcome

emmrichd, 2021-04-29:
Hello realfire,

your meter does not "speak" SML. It speaks what is often referred to as "D0" or IEC62056-21, which is roughly speaking an ASCII version of SML, which SMLReader does not understand, yet. Depeding on what you are trying to achieve, a workaround might be Tasmota or ESPLink.
Regards, Dieter

realfire23, 2021-04-29:
> Hello realfire,
>
> your meter does not "speak" SML. It speaks what is often referred to as "D0" or IEC62056-21, which is roughly speaking an ASCII version of SML, which SMLReader does not understand, [...]

where did you get that info ? according the specs I have it uses SML. Furthermore Volkszaehler and homemmatic integrated the meter already via sml ( as far as I understood of course)

emmrichd, 2021-04-30:
Hello,
you might be right. with the second look, I also saw the SML note in the manual.
I guessed because according to this list: [https://www.loxone.com/enen/info/smart-energy-meter-list/](url) most of Landis+Gyr meters don't use SML.