mruettgers/SMLReader #17: ED300L - Jumping back to reduced data set after 120 sec

trilkin-github, 2021-05-05:
Hello,

first of all thank you for providing SMLreader to the community! Worked fine immediately with the provided instruction for platform.io and my NodeMCU v2 + Hichi IR ttl!

I have the following issue: My EMH ED300L is providing the full data incl. power after entering the PIN. After 120 sec it jumps back to the reduced data set. After contacting my provider they confirmed that the option to activate full dataset constantly is not available due to an old software version which they do not want to update. 

Now my thought was: Why not let my NodeMCU enter the PIN every ~120 sec. via the Hichi IR ttl. 
It would require the transmitting IR-LED to be activated in a certain sequence (2x short, then depending on PIN code).

Do you have any recommendation on how to integrate that into the code? 
Rough idea: 
- In Config file define PIN for sending IR
- In Config file define Pin-Code (array?)
- In Config file define time between Pin entering
- In main loop check time_since_last_pin_entering
- pin entering function: 2x short activation (0,5 sec on, 0,5 sec off) of IR led to get to PIN entering (see also [here](https://www.mit-n.de/fileadmin/user_upload/Dateien/Messwesen/Messwesen_Strom/ED300L-G_ED300S-G-BIA-D-1.00.pdf)   page 30), then enter pin based on code in array

Would appreciate your thoughts and help!

mruettgers, 2021-05-05:
Hi trilkin,

thanks for using SMLReader.
I'm currently working on a new version of SMLReader that will allow the reading heads to be configured via a web interface.
Regarding to your issue I'm considering to add an option to set a PIN for being transmitted to the meter once or periodically depending on the configuration.
But I guess, we then need a further option to define the sequence for entering the PIN, because that could be different for each meter.

Regards,
Michael

trilkin-github, 2021-05-06:
Hi Michael,
that sounds promising!
Yes the sequence might be different per meter. First thoughts:
1) let people "record" the sequence via the app by toggling the lamp/led on/off. 
- pro: By that it would be independent from the meter sequence.
- con:

trilkin-github, 2021-05-10:
Hi Michael,
I found out on the weekend that once PIN is entered it is enough to just 'flash' every couple of seconds to go the next menu entry and by that avoid that the meter is switching back to reduced dataset. So I added into the main loop a regular flash of the IR led and it worked.
I can now receive the data constantly to my Home assistant...
Greetings
Johannes

mruettgers, 2021-05-10:
Ok cool.
Thanks for the feedback!

Michael

panhans, 2021-09-28:
> Hi Michael, I found out on the weekend that once PIN is entered it is enough to just 'flash' every couple of seconds to go the next menu entry and by that avoid that the meter is switching back to [...]

So I have the same problem. Did you connect a LED and stick it over the sensor for the flashlight or does it even work when the ir-led of the optical rs232 interface flashes periodically?

trilkin-github, 2021-09-28:
I use this IR read & write: https://www.ebay.de/itm/ttl-ir-lesekopf-lese-schreib-Kopf-EHZ-Volkszaehler-Hichi-Smartmeter-/313460034498?mkcid=16&mkevt=1&_trksid=p2349624.m46890.l49286&mkrid=707-127634-2357-0

[image]

Just added the ir led flash to main loop. Once you enter the pin and stick the IR led on the optical interface it then continuously flashes and by that "jumps" through the menu avoiding the timeout.

panhans, 2021-09-28:
Oh, I have the same head but the ED300S instead of ED300L. The L-version has the optical sensor build in the optical rs232 interface. The s-version has a seperate sensor. So if I also wont get a firmware update I have to put a LED before the seperated sensor.
For me the inf-menu entry is missing. There you're able to disable the reduced data set send over the optical rs232 interface even when your device is locked.

panhans, 2021-10-02:
This is my solution. Ugly but it works. XD

[image]

trilkin-github, 2021-10-02:
Beautiful - congrats!