mruettgers/SMLReader #18: AardvarkTCP.h de

steffenvogelsang, 2021-05-16:
AardvarkTCP.h dependency

The AardvarkTCP.h dependency is not a published library, or it is not anymore. I found it on Github, but it would be nice if it was maintained by PIO.

mruettgers, 2021-05-16:
Assuming you are talking about "ESP AsyncTCP", it indeed looks like a typo in the `platformio.ini` file. Will have a look at it tomorrow.
In the meantime you can try to use the GitHub URL: https://github.com/me-no-dev/ESPAsyncTCP

mruettgers, 2021-05-16:
Oh sorry. I never heard of AardvarkTCP.h before, but it seems to be related to the MQTT client lib. Will have to investigate it further.

steffenvogelsang, 2021-05-16:
You are right, it is part of "Pangolin MQTT Client", not from your code directly. I have not found Pangolin as an available lib in PIO, but I'm new in the PIO community, maybe I have missed somethink. 

Thanks for very fast response :-)

mruettgers, 2021-05-17:
> You are right, it is part of "Pangolin MQTT Client", not from your code directly. I have not found Pangolin as an available lib in PIO, but I'm new in the PIO community, maybe I have missed somethink.

That might be the reaseon why I added the library by its GitHub-URL in the `platform.io` file.
Is it possible, that your PIO is not able to download it because there's no Git-Client available?

BargMR, 2021-05-22:
Hi Michael (hoffe Dutzen ist okay - achja und deutsch schreiben auch ;-)

Vorab: **Super Arbeit - deine vorkompilierte BIN (V2.2.0) läuft wie geschmiert!**

Ich wollte aber gerne für den zweiten Zähler noch einen weitern Lesekopf anschließen und habe mich ans selber kompilieren gemacht.
Leider habe ich aber das gleiche Problem wie Steffen mit der PangolinMQTT lib. Ich kann aber sagen, dass alle libraries importiert werden. Auch habe ich in einem zweiten Versuch alle libs heruntergeladen. Beides mal das selbe Problem ...  nach der Ausgabe im Terminal zu schließen sind wohl Fehler in den libs selber. 

Deine V2.1.6 lässt sich problemlos kompilieren - aber da war ja das Sendeproblem von zwei Quellen an den MQTT noch drin.

Hier ein Terminal-Ausschnitt (nur env:d1_mini):
[image]

Den kompletten Terminaloutput häng ich als TXT an:
[terminalausgabe 2021-05-22 13h06.txt](https://github.com/mruettgers/SMLReader/files/6526290/terminalausgabe.2021-05-22.13h06.txt)

Sollten die Probleme bei dir nicht replizierbar sein, wäre es toll, wenn du noch eine Version2.2.0 mit zwei Sensoreingängen kompilieren würdest oder du deine Idee implementierst, dass man Anzahl der Sensoren und Intervall in denen sie senden auf der Installations-Webseite wählen kann - das wäre mega!

Viele Grüße aus Hessen,
Michael

ENGLISH by google-translate:
First of all: ** Great work - your precompiled BIN (V2.2.0) runs like clockwork! **

But I wanted to connect another reading head for the second counter and I started compiling it myself.
Unfortunately I have the same problem as Steffen with the PangolinMQTT lib. But I can say that all libraries are imported. I also downloaded all libs in a second attempt. Both times the same problem ... to close after the output in the terminal are probably errors in the libs themselves.

Your V2.1.6 can be compiled without any problems - but there was still the problem of sending two sources to the MQTT.

Here is a terminal excerpt (only env: d1_mini):
! [image] (https://user-images.githubusercontent.com/47054535/119224329-2e872380-bafe-11eb-95e1-060dc1ab4cb0.png)

I attach the complete terminal output as TXT:
[terminalausgabe 2021-05-22 13h06.txt] (https://github.com/mruettgers/SMLReader/files/6526290/terminalausgabe.2021-05-22.13h06.txt)

If the problems cannot be replicated for you, it would be great if you would compile a version 2.2.0 with two sensor inputs or you could implement your idea that you can choose the number of sensors and the interval at which they send on the installation website - that would be awesome!

Many greetings from Hessen,
Michael

mruettgers, 2021-05-22:
Hallo Michael,

ich konnte das Problem gerade replizieren, nachdem ich bei mir die lokal gecachten Libs mal gelöscht habe.
Anscheinend gibt es im aktuellen Entwicklungs-Branch änderungen, die nicht kompatibel mit älteren Versionen sind.
Ich habe die PangolinMQTT-Version jetzt in der ` platform.io`  auf v1.0.0 fixiert, konnte es aber mangels Zeit noch nicht testen.
Kompilieren ließ sich SMLReader damit wieder.
Falls jemand testen möchte, nur zu, ich werde das bei mir heute Abend nachholen, wenn die Kinder schlafen ;-) .

Viele Grüße,
Michael

BargMR, 2021-05-22:
Super, Danke!

[image]

Kompilieren hat gefunzt ... lade es gleich auf meinen D1mini und checke den MQTT.

mruettgers, 2021-05-22:
Ok cool. Dann schließe ich das Issue.
Und danke für den Kaffee :-)!

closed 2021-05-22 by mruettgers

mruettgers, 2021-05-22:
@steffenvogelsang Dir auch Danke :-)