mruettgers/SMLReader #21: Add a blacklist for OBIS Keys

benniju, 2021-05-23:
First of all thank you very much for the Code, it works very nice!

It would however be useful if you could add a Blacklist for OBIS Keys in the Config, that will not be published to MQTT. 
For examlpe my Smartmeter (an EMH Gen-k) outputs a OBIS Value for the Private Key, which is in HEX format but because it is send in ASCII it wastes 143 Bytes (48 Bytes in HEX) each time the Smartmeter sends new values and this is quite unnecassary.

The OBIS Code is `129-129:199.130.5`, but I think it would be better to have an option with the abilitiy to add multiple keys in the Config.

The different OBIS Codes are mentioned in [the manual](https://emh-metering.com/wp-content/uploads/2020/08/eHZ-K-BIA-D-1-20.pdf) on the pages 14-15.

mruettgers, 2021-05-23:
@benniju The successor of SMLReader will give you the possibillity to selectively enable or disable OBIS identifiers for being published via a web interface. In the meantime you can set `numeric_only`  to `true` to prevent the ASCII values from being published.

benniju, 2021-05-23:
Ok, thanks!

mruettgers, 2021-05-29:
Closing for now.
Will be implemented in the upcoming version.

closed 2021-05-29 by mruettgers