mruettgers/SMLReader #22: Wlan will not work

FluppFish, 2021-05-27:
Hello Michael,
thank you for your cool Project!
But i have a problem with my wlan credentials.

SSID: Aquarium
PASS: X{dfe5%-+xI_2%s"fd0v2}=$9{]z#S?0X@8qERT54.s6G5W>\GhB*(&$?j6=2fw

SMLReader will not login to my accesspoint and do not publish something on my mqtt server.

If i use easyer login data like
SSID: Aquarium
PASS: asd123ASD
it will work.

Do you have a solution for that?
I tried to escape the " and the \ 
X{dfe5%-+xI_2%s\"fd0v2}=$9{]z#S?0X@8qERT54.s6G5W>\\GhB*(&$?j6=2fw
but this also will not work...

Thank you for your help...
Best regards
Markus

benniju, 2021-05-27:
First of all, I hope that this is not you real password, you should really remove that if that's the case. Maybe you could try enabling the DEBUG Mode and look at the output?

Possibly the password is just too long for the default ESP8266 WiFi library?

FluppFish, 2021-05-28:
For sure this is not my real wlan password!
Only the special charters are the same...

The length of the password shouldn't be the problem.
I thested this and it worked with hard coded credentials. I only had to escape the " and the \.
https://github.com/MakeMagazinDE/Node-RED-Special/blob/master/02-MQTT/mqtt_pub.ino

syssi, 2021-05-28:
It could be an issue of the configuration layer (`IotWebConf`). May be the POST payload of the configuration form isn't decoded properly. Could you try to print the stored password as debug output at the serial line? Do you see a difference between the input and output?

ostpol, 2021-05-28:
As a workaround, try this: [http://jorisvr.nl/wpapsk.html](http://jorisvr.nl/wpapsk.html)

FluppFish, 2021-05-28:
Leider bin ich nicht so fitt.
Anbei mal der debugoutput beim start:
`{d<\0>dœŸ|<\0>Œ$à|<3><\f><\f><\f>„<4>lì<\f>c|Žƒ<3>ì<19>“{›#Œ<\f>b„ûogždooœãä<4>b<28>pì‡l{lslxògà<16><3><4><\f>‚<\f>d<4>Œœ<\f><4><4>c<\f>gã|<3>dä<\f><\f>Žb„ûggï<\0>d„‡l`<3><27><18>gn<4>l`<2><7><3>n{›’g<4><4>c<\f>œ<15>d<14>{›Ûg<\f><4>c<\f>œ<15>dÜc{c<\f><\f>ŽàãÄÛ`<3>üƒoœ<3>Setting up 1 configured sensors...<\r><\n>Initializing sensor 1...<\r><\n>Initialized sensor 1.<\r><\n>State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.<\r><\n>Sensor setup done.<\r><\n>Setting up WiFi and config stuff.<\r><\n>Adding parameter MQTT server.<\r><\n>Adding parameter MQTT port.<\r><\n>Adding parameter MQTT username.<\r><\n>Adding parameter MQTT password.<\r><\n>Adding parameter MQTT topic.<\r><\n>Config size: 685<\r><\n>Loaded config 'iwcThingName'= 'SMLReader'<\r><\n>Loaded config 'iwcApPassword'= <hidden><\r><\n>Loaded config 'iwcWifiSsid'= 'Aquarium'<\r><\n>Loaded config 'iwcWifiPassword'= <hidden><\r><\n>Loaded config 'iwcApTimeout'= '30'<\r><\n>Loaded config 'mqttServer'= 'meinmqqtserver.de'<\r><\n>Loaded config 'mqttPort'= '1883'<\r><\n>Loaded config 'mqttUsername'= 'FluppFish'<\r><\n>Loaded config 'mqttPassword'= <hidden><\r><\n>Loaded config 'mqttTopic'= 'aquarium/smartmeter'<\r><\n>Setup done.<\r><\n>State changing from: 0 to 2<\r><\n>Setting up AP: SMLReader<\r><\n>Use password: <hidden><\r><\n>AP IP address: 192.168.4.1<\r><\n>State changed from: 0 to 2<\r><\n>Connection to AP.<\r><\n>Request for www.msftconnecttest.com redirected to 192.168.4.1<\r><\n>Request for detectportal.firefox.com redirected to 192.168.4.1<\r><\n>Request for www.msftconnecttest.com redirected to 192.168.4.1<\r><\n>`

FluppFish, 2021-05-28:
I think i found the problem.
> https://s20.directupload.net/images/210528/thotysh7.jpg

Max size of the html password field is 32 charters...

And now - any idea how to solve it?

syssi, 2021-05-28:
I we assume the library is able to store more than 32 chars you could use the developer tools of your browser to manipulate this value. Remove the parameter or increase the maxlength. Fill the form and press submit. If 32 chars is a hard limit the POST endpoint must be checked / changed / fixed (if possible).

mruettgers, 2021-05-28:
This isse seems to be caused by a limit in the IotWebConf library, that is used for WiFi configuration:
https://github.com/prampec/IotWebConf/blob/5a8ba88a0bca5be5771ffa939a252edc021bde02/src/IotWebConf.h#L39

You might be able to override this limit this way in `main.cpp`:

```
# define IOTWEBCONF_PASSWORD_LEN 64
#include <IotWebConf.h>
```

FluppFish, 2021-05-28:
It will no work.

I downloaded the latest src.
Added the "#define IOTWEBCONF_PASSWORD_LEN 65" above the #include of the IotWebConf.h
Used the .\d1_mini_debug\firmware.bin for program the esp8266.

`sd<\0>lœŸ|<\0>Œdà|<3><4><\f><\f>Œ<4>dä<\f>c|Çƒ<3>ä<27>’s“cŒ<\f>cŒóooŸl'nœâä<\f>c<28>p„l{d;lpûgà<24><3><4><4>ƒ<\f>l<4>„œ<\f><\f><\f>#<4>oâ|<2>lä<4><\f>‡cŒóooî<\0>l„$`<3><27><27>og<\f>d`<3><15><2>gsÇ“›o<\f><\f>c<4>œ<14>d<7>s“›o<\f><\f>c<4>œ<7>lœbsc<4><\f>‡àã„›`<3>üƒoœ<2>Setting up 1 configured sensors...<\r><\n>Initializing sensor 1...<\r><\n>Initialized sensor 1.<\r><\n>State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.<\r><\n>Sensor setup done.<\r><\n>Setting up WiFi and config stuff.<\r><\n>Adding parameter MQTT server.<\r><\n>Adding parameter MQTT port.<\r><\n>Adding parameter MQTT username.<\r><\n>Adding parameter MQTT password.<\r><\n>Adding parameter MQTT topic.<\r><\n>Config size: 685<\r><\n>Loaded config 'iwcThingName'= 'SMLReader'<\r><\n>Loaded config 'iwcApPassword'= <hidden><\r><\n>Loaded config 'iwcWifiSsid'= 'Aquarium'<\r><\n>Loaded config 'iwcWifiPassword'= <hidden><\r><\n>Loaded config 'iwcApTimeout'= '30'<\r><\n>Loaded config 'mqttServer'= 'meinmqttserver.de'<\r><\n>Loaded config 'mqttPort'= '1883'<\r><\n>Loaded config 'mqttUsername'= 'FluppFish'<\r><\n>Loaded config 'mqttPassword'= <hidden><\r><\n>Loaded config 'mqttTopic'= 'aquarium/smartmeter'<\r><\n>Setup done.<\r><\n>State changing from: 0 to 2<\r><\n>Setting up AP: SMLReader<\r><\n>Use password: <hidden><\r><\n>AP IP address: 192.168.4.1<\r><\n>State changed from: 0 to 2<\r><\n>Did not receive an SML message within 30 seconds, starting over.<\r><\n>State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.<\r><\n>State changing from: 2 to 3<\r><\n>Connecting to [Aquarium] (password is hidden)<\r><\n>State changed from: 2 to 3<\r><\n>Did not receive an SML message within 30 seconds, starting over.<\r><\n>State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.<\r><\n>Giving up.<\r><\n>State changing from: 3 to 2<\r><\n>Setting up AP: SMLReader<\r><\n>Use password: <hidden><\r><\n>AP IP address: 192.168.4.1<\r><\n>State changed from: 3 to 2<\r><\n>Start sequence found.<\r><\n>State of sensor 1 is 'READ_MESSAGE'.<\r><\n>End sequence found.<\r><\n>State of sensor 1 is 'READ_CHECKSUM'.<\r><\n>Message has been read.<\r><\n>State of sensor 1 is 'PROCESS_MESSAGE'.<\r><\n>Message is being processed.<\r><\n>SML file (1 SML messages, 42 bytes)<\n>SML message  101<\n>OBIS data<\n>State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.<\r><\n>`

FluppFish, 2021-05-29:
> Could you try to print the stored password as debug output at the serial line?

Could you please explain me how to do this?

benniju, 2021-05-29:
Well, I think it's supposed to be in the debug output that you have posted before

syssi, 2021-05-29:
You could define `IOTWEBCONF_DEBUG_PWD_TO_SERIAL`: https://github.com/prampec/IotWebConf/blob/v2.3.3/src/IotWebConf.cpp#L913-L920

FluppFish, 2021-05-29:
I tested it - there is no output.
And it is still not working.
Giving up for the moment - have to think in change my AP password...

syssi, 2021-05-29:
What did you do in detail? May be there are some more steps required to enable the debug mode of the underlying library:

https://github.com/prampec/IotWebConf/blob/v2.3.3/src/IotWebConf.h#L61-L74

mruettgers, 2021-05-29:
Side note:  
In the upcoming version of SMLReader  I will use https://github.com/khoih-prog/ESPAsync_WiFiManager instead of IotWebConf. Then there won't be a limit to 32 chars by default anymore.

FluppFish, 2021-05-30:
What do you think - when the first beta will be available for your new version?

FluppFish, 2021-05-31:
Is a known bug.
https://github.com/prampec/IotWebConf/issues/62
https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#compile-time-configuration

-> Issue could be closed from my side...

closed 2021-05-31 by FluppFish