mruettgers/SMLReader #23: [Feature Request]  MQTT Topic No Readout

Marv2190, 2021-05-29:
Hi,
thanks for your Project. After i finally got it to compile (the release binary is a little older ;) ) and it worked, I 

After I finally managed to compile it  (the release binary is a little older ;)  and it worked, I realized that a function to monitor the Device (Readout) would be nice.
Could one also set a timeout from reading the SML telegram via MQTT?

The LWT is good if the ESP dont work anylonger, but dont tell anything about the Meter.

So over one minute no new telegram -> error.

I saw in the code 👍 `DEBUG("Did not receive an SML message within %d seconds, starting over.", READ_TIMEOUT);`
but I have no clue how to publish this in MQTT ( A simple "error" "noerror" would be better anyway).

Maybe thats an easy one for you to add.

Danke

syssi, 2021-05-29:
In my case Home Assistant consumes the MQTT messages and the `last_updated` timestamp of the sensor gets updated on each message. This is all I need to trigger automations.

Marv2190, 2021-05-29:
Yeah, that would be a work around. I have no last_updated (using NodeRed), but i could use a trigger with delay i think.
Maybe for SMLreader v2 this is an option.

mruettgers, 2021-05-29:
> Yeah, that would be a work around. I have no last_updated (using NodeRed), but i could use a trigger with delay i think.
Or you could implement your own ` last_updated`  using a flow variable, that is being monitored by a periodic trigger.

> Maybe for SMLreader v2 this is an option.
That won't be a a big deal to implement and I already thought about publishing info/debug messages via MQTT.
Right now debug messages are being published as server sent events in the upcoming version.

Marv2190, 2021-05-29:
I just started with Node-red (because its easy to feed mqtt to Homematic). So iam not deep into it. 
In the moment a tried the trigger node (with delay that set it self back with a new message), and it works pretty well. But Info/debug would be a nice to have, ofc. And most of them i found in the code - so its just a giveover to mqtt. :)

Anyway, thanks for this project. 

Grüße nach Aachen und Bonn :)

trombose009, 2021-07-09:
Does MQTT also work if no MQTT user and password is set?  In my environement no mqtt user is necessary.

Marv2190, 2021-07-09:
Yes, thats how i use it