mruettgers/SMLReader #24: Great Work

trosenda, 2021-05-31:
thanks for the code, it helped me to spot some strange electricity usage in my house.
any idea when the sleep mode will be implemented ? 
my installation had to run on a power bank as there is no socket close to the meter.
I would be fine with one telegram every 300 seconds.

thanks !

FluppFish, 2021-05-31:
check config.h
.interval = 300 is maybe the solution you are looking for...

mruettgers, 2021-06-02:
Right now I don't have a clue when I will have the time to implement this feature.
Configuring an interval won't help here, because the ESP8266 keeps running and thus consuming energy.
Regarding the deep sleep feature I was thinking to switch over to the ESP32, because the power consumption in deep sleep is less than the one of the ESP8266 and because there is much more RTC memory available that could be used during deep sleep to cache measurements before sending them batched via wifi.

trosenda, 2021-06-04:
That sounds very advanced!
I was thinking of an easier (less detailed) way, really just waking up for
like 10 seconds, going to sleep for 300. 1.8.0/2.8.0 is good enough for my
use case, don't need much more

Thanks for the reply!

trosenda, 2021-06-12:
quick test only, seems it wasnt it. any pointers where to start ?

mruettgers, 2021-06-12:
It might work this way, but untested:
https://github.com/mruettgers/SMLReader/commit/35d91de730de1e13b46e80f6d27cc3d9788ddaaf

But in order to make timer based wakeup work after putting the device into deepsleep you have to connect GPIO16/D0 to RST.

trosenda, 2021-06-13:
Thank you.
It goes into sleep and comes back (big challenge was to find a powerbank which doesnt turn off because off low current during sleep)
I am unfortunately only receiving the following :

`
2021-06-13 11:47:48 MQTT2_DEVICE MQTT2_esp8266_02A548 info: Hello from 0002A548, running SMLReader version 2.2.0.
2021-06-13 11:47:48 MQTT2_DEVICE MQTT2_esp8266_02A548 LWT: Online

2021-06-13 11:52:53 MQTT2_DEVICE MQTT2_esp8266_02A548 info: Hello from 0002A548, running SMLReader version 2.2.0.
2021-06-13 11:52:53 MQTT2_DEVICE MQTT2_esp8266_02A548 LWT: Online

2021-06-13 11:57:54 MQTT2_DEVICE MQTT2_esp8266_02A548 info: Hello from 0002A548, running SMLReader version 2.2.0.
2021-06-13 11:57:54 MQTT2_DEVICE MQTT2_esp8266_02A548 LWT: Online`

mruettgers, 2021-06-13:
Sounds like the ESP ist going to sleep to fast. Try to add a delay before the sleep command  in a non-blocking manner (Short delays of i.e. 10ms together with a yield call within a loop).

trosenda, 2021-06-14:
Excellent, coming close.
running since a couple of hours, may need to increase the loop a bit, since I have 56 Info vs 44 1-0_1.8.0_255_value readings. so sometimes it sleeps to quick. 300 a good value or any other recommendation ?

but will have to wait until tomorrow before I can bohter the handyman for roomkey again.

trosenda, 2021-06-20:
The wemos with powerbank via usb was (obviously) drawing to much power.
did now change to an ESP8285 powered by 4AA batteries (2S,2P)
10k or 4.7k pullup did not work, I went with 1k again.

mruettgers, 2021-06-20:
In case it is still drawing too much power it might be an option to power the circuit with a LiFePO4 battery. Thus you can omit any LDO in between. In another project I'm running an ESP32 drawing about 10 μA in deep-sleep while directly connected to a LiFeP04 cell. With an LDO the circuit was drawing about 35 μA in deep-sleep.

trosenda, 2021-06-21:
LiFePO4 would come by the cost of chargers etc. 
Let's see how long it will last on AA, I will report. BTW, no LDO, connected batteries directly to +3.3.
Unfortunately my DMM dont allow for autorange between µA and mA, so I couldn't accurately measure.

mruettgers, 2021-06-21:
Ah, I see.
Btw., I'm using https://lowpowerlab.com/guide/currentranger/ for measuring, which is really cool.
The serial logging option combined with auto-ranging allows me to write directly into an InfluxDB database and to analyze with Grafana.

trosenda, 2021-06-22:
that's a very nice device. I have to resist spending so much money ;)

trosenda, 2021-07-02:
The first set of batteries (4x Varta industrial AA) lasted from 20.6-2.7. not as long as I hoped for. I might need to invest into something like power ranger. My poor man measurement (guess work, really) showed 140micro amps during sleep.

mruettgers, 2021-07-08:
That sounds like the LDO is drawing too much power during deep sleep.
And that was one of the reasons I'm working with a LiFePo4 cell now (~7 µA in deep sleep for an ESP32).

PS: I renamed the topic because other people might be interested in battery powered operation, too.

trosenda, 2021-07-08:
WeMos lite. Batteries connected directly to 3.3.
The second set lastet only till today, which is only 6 days. Didn't check yet (room key, handy man, etc), just saw I don't see any messages no more. 
Will see over the weekend if I get plain 8285, or invest into better current meter.

SBajonczak, 2021-07-08:
Hi I use a lipo for another Projekt on a wemo d1 mini. This ging to sleep for the maximum sleep time (I think, it was 30 Minutes). I use this setup for measuring a beehive with an hx711 and several temp Sensors,but the runtime is about a month with a 3,7v lipo. In did Not Do any solder magic. But there are several "China" stuff that contains a uart in it. This Part consumes the most currency, and that was my Prior fail into this.

I am currently on my holdiay so that I can contribute only when I am at home.

Maybe you can attach a Puffer battery like the lipo, a cn365 (Red one) Solar charger and a Solar cell. Then it can run nearly forever?

lal12, 2022-02-02:
Did you experiment with the wifi sleep modes? I only tried them with ESP32s before, where they are decreasing power usage significantly. 

There is WIFI_PS_MIN_MODEM, where the modem awakes every DTIM period and there is WIFI_PS_MAX_MODEM which sleeps longer (as configured). The latter one can cause issues with packet loss (which should be handled by the TCP layer) and in rare cases there are problems with some APs though I didn't had any issues with that. But even the MIN mode already decreases power usage quite a bit.

The WiFiClient.setSleep* methods are usable for that.