mruettgers/SMLReader #26: How to bring SMLReader into existing network

Pewibe, 2021-06-24:
Help is needed!
1. My hardware is a Sonoff Basic R3. By using Tasmota the device is running fine.  
2. I flashed both pre-fabricated versions SMLReader_D1mini_v2.2.0[_debug].bin.
3. I get the configuration page and entered my data. Problem is the field "AP password" because there is no clear description for the necessary content. I entered "ASDFqwer1234".
4. I received the message "Configuration saved ...".
5. The device stay in access point mode and do not participate in configured home network. Does it happen automatically or do I have to do something for this next step? I know that the two Wifi parameters are important for this step. But I'm pretty sure to do it correct because with Tasmota software on this device everything works.
Thanks in advance!

trosenda, 2021-06-24:
AP Password is the WiFi Password when in AP mode (your device unable to connect to given wifi sssid/passwd)
make sure you enter wifi ssid, wifi password, ap password.
save
powercycle

if it doesnt connect, check your dhcp server or flash the debug and share output of serial console.

Pewibe, 2021-06-24:
@trosenda Thanks for your answer but I'm still missing hints to go further.
1. You explained the usage of AP password but not the syntax. Except minimum 8 characters there is no clear documentation for the necessary types of characters. Or easily say if my used password "ASDFqwer1234" should work.
2. DHCP etc. is double checked by using Tasmota software (everything incl. SSID/PW works fine).
3. Debug output shows:
_Setting up 1 configured sensors...
Initializing sensor 1...
Initialized sensor 1.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Sensor **setup** done.
Setting up WiFi and config stuff.
Adding parameter MQTT server.
Adding parameter MQTT port.
Adding parameter MQTT username.
Adding parameter MQTT password.
Adding parameter MQTT topic.
Config size: 685
Wrong config version.
Missing or invalid config. MQTT publisher disabled.
Setup done.
AP password was not set in configuration
State changing from: 0 to 1
Setting up AP: SMLReader
With default password: <hidden>
AP IP address: 192.168.4.1
State changed from: 0 to 1
Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'._
No idea about "wrong config version" and why my 12 chars test AP password is not used.

mruettgers, 2021-06-24:
Could you please post the debug output after reboot?
Before you edited your message I noticed something like "Wrong config version".
It looks like the Sonoff Basic R3 is driven by a ESP8255 chip.
The library used for configuration (IotWebConf) is using EEPROM for storage, maybe there is an issue with the EEPROM of the ESP8255?
Unfortunately I don't have any ESP8255 device here to test with.

mruettgers, 2021-06-24:
"Wrong config version" indicates, that the version of the config in EEPROM does not match the one that has been written previously, or that there is no config found at all.

Pewibe, 2021-06-24:
I added the missed lines to the previous debug output. Sorry for the mess of the font size. I'm sure that you are right concerning 8255. So I cannot use the prefabricated .bin and have to compile.

closed 2021-07-03 by mruettgers