mruettgers/SMLReader #27: MQTT bei IObroker - Strukturfehler bei Übertragung

Basti-RX, 2021-07-02:
Hallo,
wäre es möglich folgendes anzupassen?

```
smartmeter/mains/sensor/1/obis/1-0:1.8.0/255/value 3546245.9
smartmeter/mains/sensor/1/obis/1-0:2.8.0/255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.1/255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:2.8.1/255/value 13.2
smartmeter/mains/sensor/1/obis/1-0:1.8.2/255/value 3546245.9
smartmeter/mains/sensor/1/obis/1-0:2.8.2/255/value 0.0
smartmeter/mains/sensor/1/obis/1-0:16.7.0/255/value 451.2
```
Also den Pfad der Übertragenden Daten? Bei IOBroker nimmt er leider jeden . als neuen Unterordner für den Datenpunkt. 
Das heißt er erstellt eine ewiglange Struktur. Habe mal ein Bild angehangen.
[image]

mruettgers, 2021-07-02:
Hallo,

beeinträchtigt das die Funktionalität oder ist es nur ein Schönheitsfehler in ioBroker?
In meinen Augen ist es ein Fehlverhalten von ioBroker, anzunehmen, dass ein `.` im MQTT-Topic als Trennzeichen zu interpretieren ist.
`A-B:C.D.E*F` ist ja auch das offizielle Format für die Notation der OBIS-Kennzahlen (das `*` habe ich wegen Problemen mit ioBroker bereits durch `/` ersetzt.)

Allerdings tue ich mich gerade etwas schwer, auf Grund eines ioBroker-Problems den ganzen OBIS-Identifier anzupassen, und wenn, welches Format wäre denn geignet?
Zumal dann jeder Nutzer beim nächsten Update seine Skripte/Logiken auf die neuen MQTT-Topics anpassen müsste.

Da ich aber auch gerade kaum Zeit für Anpassungen habe, ist es am einfachsten, wenn Du das an dieser Stelle Deinen Wünschen entsprechend anpasst:

https://github.com/mruettgers/SMLReader/blob/develop/src/MqttPublisher.h#L83

Für den Nachfolger von SMLReader behalte ich den Feature-Request aber mal im Hinterkopf.

Viele Grüße,

Michael

Basti-RX, 2021-07-04:
Danke werde ich mal anpassen. Hatte es schon gefunden gehabt nachdem ich den Issue erstellt hatte

closed 2021-07-07 by mruettgers