mruettgers/SMLReader #28: display

trombose009, 2021-07-09:
Hey, 
sorry, I do not not if it's possible or how you can write personal messages to other people here. 
Your project is great and I will adapet it to my grafana/inluxdb environement. 
My question: Is there any similar project, which works without a database? I am interested just in showing the actual current consumption on a display with an esp32/8266 . 
Or is there an ino-file which i could edit to add a display by myown? I am not able to work with the bin-files in that way. Or maybe it is easy to cenvert them?
omg! I am such a noob ;)

benniju, 2021-07-09:
The source code is available in the repository. You have to edit and compile it with platformio.

About the display: Do you want to show the data on the same device that records it or on a remote device?

A database is not necassy, just a MQTT server is enough if you don't want to store the data.

trombose009, 2021-07-09:
Ah, ok. You are right. I should just receive the mqtt msgs on a additional device to display it there.

closed 2021-09-22 by mruettgers