mruettgers/SMLReader #29: Question: OTA update possible?

derandiunddasbo, 2021-07-17:
Is it possible to use the OTA update provided by IOTWebConf to upload new/self compiled firmware versions?

mruettgers, 2021-07-19:
OTA updates should already work: http://[your-ip]/firmware (or use the link on the start page).

derandiunddasbo, 2021-07-20:
Ok, thanks for the clarification. I just wanted to make sure, I won't "brick" my device by attempting an OTA update. It's sitting at a quite inaccessible place and I want to avoid having it to update over a wired connection.

closed 2021-07-20 by derandiunddasbo

derandiunddasbo, 2021-07-22:
Just wanted to give a quick follow up: Building the firmware and the OTA update worked fine. 

After reboot I was a bit confused at first, because I couldn't reach the web config page anymore and it didn't publish anything to the mqtt server. But then I realized, that the devices captive portal was active again and I just had to reconfigure its WiFi and mqtt settings.