mruettgers/SMLReader #30: WIFI config not saved correctly

btmerz, 2021-08-02:
Hello,
after uploading the project I've connected to the AP and entered the WIFI values, and MQTT password and so on. All fields were correctly filled and I received the message "Configuration saved. Please disconnect from WiFi AP to continue!". After Restart of the ESP I received the following Info in the debug window:
Initializing sensor 1...
Initialized sensor 1.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Sensor setup done.
Setting up WiFi and config stuff.
Adding parameter MQTT server.
Adding parameter MQTT port.
Adding parameter MQTT username.
Adding parameter MQTT password.
Adding parameter MQTT topic.
Config size: 749
Wrong config version.
Missing or invalid config. MQTT publisher disabled.
Setup done.
AP password was not set in configuration
State changing from: 0 to 1
Setting up AP: SMLReader
With default password: <hidden>
AP IP address: 192.168.4.1
State changed from: 0 to 1
Connection to AP.

And I can't connect to the ESP by WIFI. What could be wrong there?
Thanks for your help!

mruettgers, 2021-08-02:
Hey,

can you provide some details about the character lengths (and the use of special chars) of the configuration params? Especially the one of the passwords (WiFi/MQTT)?
Thanks.

Michael

btmerz, 2021-08-05:
Hi,
my wifi password consists of 20 numbers and the MQTT password has 11 digits, with upper and lower case characters, one number and a !
But with the precompiled version it works!?

KShips, 2021-08-05:
Did you add a AP password whens setup? I had same issue and then realized password couldn't be blank. Adding password fixed issue.
Using V2.2.0

btmerz, 2021-08-05:
Yes, I added also a AP password similar to the MQTT password...

btmerz, 2021-08-08:
Are there any new ideas? The problem still exists.
All I want to change is, that i get the power consumption in kWh and noch in Wh...
But I can't get acces to the ESp with my changed version.

btmerz, 2021-08-08:
It seems, that there is another config version stored in the ESP which is not compatible with my compilation. Is there a way to completely erase the ESP memory-areas?

btmerz, 2021-08-08:
Meanwhile I have solved the problem -anyhow...
I took the firmware.bin file from the project folder and flashed it with the esptool as described in the main page. No it works...
So the problem is, that the upload tool in platformio doesn't erase epecific areas in the ESP memory.

Hmm...

mruettgers, 2021-08-08:
Hm, that sounds strange.
In case of similar issues in the future, you can use a tiny sketch to clear the EEPROM (https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/examples/eeprom_clear/eeprom_clear.ino).
As far as I know the IotWebConf library will ignore the contents of the EEPROM when the version of the config changes (here defined in `config.h` ). Maybe that would be also worth a try.

btmerz, 2021-08-08:
Now, as I have the right config running, I can upload directly from PlatformIO. I also changed the config version before, with no positive result.
Stange...

closed 2021-09-22 by mruettgers