mruettgers/SMLReader #31: Connection of a serial reading head

gitmors, 2021-08-05:
Hello,

I wanted to ask carefully if there is the possibility to operate with a serial reading head like the "Hichi IR v1.1" similar to this one described here: https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf -ttl-out

Thank you very much and I would certainly leave a nice donation if this could be implemented into the code, if it isn't already.

benniju, 2021-08-05:
Yes, I use that exact one and it works great. 
You can connect the TX pin of the head to a button and use a pullup resistor (switching between GND and 3.3V), then you can enter the PIN and navigate the menu with that.

gitmors, 2021-08-05:
Hello benniju,

Thanks for your fast reply.
I hope i understand what you mean. Please excuse me but i have never read the values ​​of an electricity meter myself that's why I ask so stupid ;). So you describe how I get through the menu of the electricity meter to use the reading head as a "flashlight" correct? And how do i have to connect the reading head to the esp8266 module?

Thank you very much

benniju, 2021-08-05:
Yes, I mean the menu of the meter. 

The head basically just outputs the (I think inverted state) of the input pin on the board (I am also not sure what RX and TX means on the board) and turns the IR LED on or off. (You can test the LED and pins by using a phone camera).

The other PIN just outputs the state of the Photodiode and this is the input you need to connect to the ESP, because this reads the serial data from the meter.
With the default configuration of SMLReader on WEMOS board, you have to connect it to the D2 Pin:
[image]

The other input of the head (the one that controls the LED) should be connected to either GND or 3.3V. Otherwise it will just randomly click through the menues of the meter.

Depending on the meter, you might also have to output actual serial data with the IR-LED, but I don't think SMLReader supports this yet. Which meter are you using?

gitmors, 2021-08-06:
Hi benniju,

many thanks for your response.
I have several meters in use. It is mainly about these three readers:

[image]
[image]
[image]
 

Briefly about my project: We have a multi-parent house and a number of readers, and I just want to have all the values at one screen. The heat meters are installed in the distribution boxes of the individual apartments. Thus, they would have to be integrated via wifi, which is available everywhere there. For the main meter distribution box in the basement, I would like to connect all readers to one raspberry pi and then be able to process all values with nodered on a separate server. So actually I only need the possibility to read out the heat meters over wifi for this project. 

Then I have this weird part here too. Maybe you can tell me quickly what this device is for:

[image]

Unfortunately, I have little experience with this matter and maybe you can recommend a heat meter that is generally well compatible and works with common communication standards?

thank you very much

mruettgers, 2021-08-08:
Hi @gitmors,

It looks like the heat meter (`Integral-MK UltraMaXX`) is talking M-Bus rather than SML.
So it defintively won't be supported by SMLReader.
There might be other libraries or projects about reading M-Bus based meters on an ESP8266.

Regards,

Michael

benniju, 2021-08-08:
The thing that you did not know what it is, is a so called Rundsteuerempfänger.

It is basically a relay that can be controller by the grid operator.

It is used for example to control different tariffs and turn on loads for example at night, when the power is cheap.

https://de.m.wikipedia.org/wiki/Rundsteuertechnik