mruettgers/SMLReader #35: no Data

Lanwolf81, 2021-08-31:
Hi,
i have an ebz dd3 smart meter [https://www.celle-uelzennetz.de/fileadmin/data/netze_documents/1_Kunde/Messwesen/eBZ_DD3_Betriebsanleitung_Kompaktzaehler.pdf](url)
and i use this BPW 40
[https://www.reichelt.de/fototransistor-npn-30v-100ma-0-1w-bedrahtet-bpw-40-p5806.html](https://www.reichelt.de/fototransistor-npn-30v-100ma-0-1w-bedrahtet-bpw-40-p5806.html)
because there is no bpw 40 in 3mm
I get only from the EPS8266 
"Hello from 00C14A9B, running SMLReader version 2.2.0."

with phototransistor can i use?
i use the front info interface on the dd3 smart meter.

Please can you help me

Lanwolf81, 2021-08-31:
my ebz dd3 has an sealed second interface on the top. so it use SML bidirectional

mruettgers, 2021-09-01:
What's the exact model of your meter? `DD3 - ... - SMZ1*`?

Lanwolf81, 2021-09-01:
Hi,
DD3 2R06 ETA - SMZ1

Jabberwock-v6, 2022-11-05:
Hi, I'm also a CUN customer with the same DD3 meter and the same problem. Have you solved it? I tired the reading head with led and transistor and just the easy transistor version as shown here. Any ideas?