mruettgers/SMLReader #38: Feature Request - Publish hassio discovery queues

Analogiker, 2021-10-17:
When using homeassistant as automation backend you can use discovery queues to publish all sensor data. Hassio will do an autodiscovery and adds all needed sensordata to its internal sensor-object database. 

example: 

`
DEVICE = {
    "name":"Pool Controler",
    "uniq_id":"pool_controler",
    "stat_t":"pool/poolcontroler",
    "val_tpl":"{{value_json.status}}",
    "dev":{
        "name":"Pool Controler",
        "sw":"0.1",
        "mf":"Raspberry, Atlas Tentacle T3",
        "mdl":"3b",
        "ids":[
            "pool_controler"
        ]
    }
}
RTD_value = {
    "name": "Pool RTD value",
    "uniq_id": "pool_rtd_value",
    "stat_t": "pool/poolcontroler",
    "val_tpl": "{{value_json.RTD.value}}",
    "dev": {
        "ids": [
          "pool_controler"
        ]
    }
}

device_map = [
    {"discovery_queue": "homeassistant/sensor/poolcontroler/device/config", "device": DEVICE},
    {"discovery_queue": "homeassistant/sensor/poolcontroler/rtd_value/config", "device": RTD_value},
]

for dev in device_map:
    client.publish(dev["discovery_queue"], json.dumps(dev["device"]))
`
This will create an device and an entity containing a temperature value in hassio. No manual configuration ist needet in hassio whith that function. The discovery data is sent one time when the app starts(mqtt setup function).

Would it be possible to get some kind of addon?

Kind Regards
Tobias

swa72, 2021-10-30:
+1
saves the user from manually adding configuration.

Skeletitor, 2021-10-30:
To be honest,

for hassio there is also an other project based on esphome. So I'll close this request here

swa72, 2021-10-30:
@Skeletitor u happen to have a link?

Skeletitor, 2021-10-30:
https://github.com/alengwenus/esphome_components

Its an addon for ESPHome
Readme ist here: https://github.com/alengwenus/esphome_components/blob/main/components/sml/README.md

closed 2021-10-30 by Skeletitor