mruettgers/SMLReader #39: wrong config version, Config-website does not save settings

moddmylife, 2021-11-03:
Hi! 
The ESP opens a Accesspoint everytime I stored the Config.

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
Setting up 1 configured sensors...
Initializing sensor 1...
Initialized sensor 1.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.
Sensor setup done.
Setting up WiFi and config stuff.
Adding parameter MQTT server.
Adding parameter MQTT port.
Adding parameter MQTT username.
Adding parameter MQTT password.
Adding parameter MQTT topic.
Config size: 685
Wrong config version.
Missing or invalid config. MQTT publisher disabled.
Setup done.
AP password was not set in configuration
State changing from: 0 to 1
Setting up AP: SMLReader
With default password: <hidden>
AP IP address: 192.168.4.1
State changed from: 0 to 1
Connection to AP.
Request for detectportal.firefox.com redirected to 192.168.4.1
Request for www.msftconnecttest.com redirected to 192.168.4.1`

mruettgers, 2021-11-09:
Hey,

what's the exact model of the ESP you are using?
Did you set an AP password (which is mandatory)?

moddmylife, 2021-11-18:
Hi. Yes I changed it. It's a wemos D1.

Cyklops66, 2022-02-06:
I think the problem isn`t a "save config" problem. The firmware doesnt accept Wifi passwords with more than 32 signs. So in the first place it is possible to think it does not save any inputs. But that isn`t the case.

mruettgers, 2022-03-17:
The new binary (2.2.1) should allow WiFi passwords with up to 64 chars.