mruettgers/SMLReader #40: Reading two SmartMeters with one ESP

T2FF, 2021-11-04:
Hello,
first of all i like your project very much and thanks for sharing it !

Now my question:
I have two smartmeters and would like to read them simultaneously via two BPW40 transistors with one ESP.
Is this possible and if so what changes in the code do I have to make and how do I connect the second transitor to the ESP ?

mruettgers, 2021-11-09:
Hey @T2FF,

multiple reading heads should work without problems.
You just have to configure the heads as described here:
https://github.com/mruettgers/SMLReader#configuration

syssi, 2021-11-09:
I'm reading 3 smartmeters with one ESP without issues.

T2FF, 2021-11-10:
okay thank you very much for the quick feedback.

Is this correct?
[image]

Because I do not quite understand the circuit. According to my understanding, if the 3.3V pin is an "ideal" voltage source, the same potential should always be present at the digital inputs. The only thing that changes in my opinion is the current flow, because the transistors are connected in parallel, but digital inputs can not measure current flow?

benniju, 2021-11-10:
In the image the two inputs are basically put in parallel, so they will always turn on and off together. You need two separate resistors for each sensor.

T2FF, 2021-11-10:
So you mean something like this ?
[image]

closed 2021-11-16 by T2FF