mruettgers/SMLReader #41: SMLReader running on Homeassistant Glow Hardware Setup?

kayloehmann, 2022-02-08:
I have built a homeassistant glow hardware setup and found out that it is actually not working for smartmeters. :-(
https://github.com/klaasnicolaas/home-assistant-glow

Would it be possible to configure the SMLReader to also work with this?
I have flashed the Wemos D1, also receiving the MQTT message. 
[image]

But no values. I also recompiled the firmware with the D6 pin.

Any ideas? Thanks for the great project!

Kay

benniju, 2022-02-08:
Which smart meter are you using?

kayloehmann, 2022-02-08:
I use this one: eHZ-IW8E2A5L0EQ2P
Should be SML ready
https://shop.voltaris.de/smart-metering/29-zweirichtungszaehler-ehz-iw8e2a5l0eq2p.html