mruettgers/SMLReader #43: Wiki is broken

PaddyCube, 2022-03-18:
Hello,
yesterday, I was able to read through your wiki but today, I'm not. It redirects to start page so it seems it is no longer available. As you did a commit yesterday, I suggest to check if you broken wiki with this.

mruettgers, 2022-03-18:
Hey,

that sounds strange because I did not use the wiki at all in the past.
Did you mean the README.md?

Michael

PaddyCube, 2022-03-18:
No I mean the Wiki. If I click on Wiki Link, it redirects me to starting page of this repo
[image]

mruettgers, 2022-03-18:
For me it looks like this after clicking the Wiki link:
[image]

Maybe you are redirected to the repo's starting page because there has no content been added to the Wiki yet?
At least I'm not aware of any content because I never used the Wiki before here.

PaddyCube, 2022-03-23:
I'm sorry, I mixed-up different repos. Right now, I'm working on a solar tracker which consumes data from smart meter with your SMLReader as well as some data from a PV inverter. It was the repo of PV inverter, which has a wiki explaining the overall structure of used MQTT protocol as well as other instructions.
I thought I've seen similar in your repo, but this isn't true.

closed 2022-03-23 by PaddyCube