mruettgers/SMLReader #44: After Update  to 2.2.0 aedes broker crashes

Snotmann, 2022-04-18:
I updated to 2.2.0 today and since then the aedes mqtt broker crashes.

Apr 18 11:41:06 raspberrypi Node-RED[18988]: 18 Apr 11:41:06 - [error] TypeError: Cannot read property 'toString' of undefined
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at Aedes.authenticate (/home/pi/.node-red/node_modules/node-red-contrib-aedes/aedes.js:172:68)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at ClientPacketStatus.authenticate (/home/pi/.node-red/node_modules/aedes/lib/handlers/connect.js:101:17)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at makeCallTwo (/home/pi/.node-red/node_modules/fastseries/series.js:150:3)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at release (/home/pi/.node-red/node_modules/fastseries/series.js:138:7)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at resultList (/home/pi/.node-red/node_modules/fastseries/series.js:125:3)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at Aedes.series [as _series] (/home/pi/.node-red/node_modules/fastseries/series.js:45:7)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at Immediate.init (/home/pi/.node-red/node_modules/aedes/lib/handlers/connect.js:83:17)
Apr 18 11:41:06 raspberrypi Node-RED[18988]:     at processImmediate (internal/timers.js:466:21)
Apr 18 11:41:06 raspberrypi systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE

mruettgers, 2022-04-19:
I will have a look at it in the next few days. I have never used Aedes before and need to get it working for further testing.

Snotmann, 2022-04-19:
Ok my fault ... the SML Reader seems to be buggy after update it to 2.2.0 so mqtt broker crashes. After i disconnect it from power for a minute and take it back to network it worked fine again with 2.2.0

closed 2022-04-19 by mruettgers