mruettgers/SMLReader #45: Sending Intervall

kitt2011, 2022-05-10:
Hello,

thanks for the programm! What is the standard sending intervall to the mqtt server and is there a way to change it?
Thank you, great work!

mruettgers, 2022-05-10:
Hey,

by default, SMLReader sends MQTT messages as often as data is received from the smart meter.
You can limit the messages by adjusting the interval value (seconds) in the config.h (https://github.com/mruettgers/SMLReader/blob/develop/src/config.h#L23).

geronet1, 2022-11-17:
Würde doch reichen wenn er nur sendet wenn sich der Wert ändert, das würde auch das Problem mit den festen Strings umgehen daß die ständig mitgesendet werden. Außerdem kann man es so einfach in die Datenbank schreiben, sonst steht dort zu oft der gleich Wert drin. Soll ich das mal integrieren? Vielleicht sogar auswählbar in der Config

mruettgers, 2022-11-17:
Gerne, PRs sind jederzeit willkommen :).
Wobei das mit dem Senden nur bei Änderung beim Auswerten des Momentanverbrauchs eher kontraproduktiv ist.
Vielleicht dann so konfigurierbar, dass es nur für bestimmte OBIS-Identifier gilt?
Beachte aber bitte, dass es bald eine Web-basierte Konfiguration gibt (https://github.com/mruettgers/SMLReader/pull/54), also würde es sich anbieten, diesen Branch dann als Vorlage zu nehmen.

geronet1, 2022-11-17:
Du meinst den Bezugswert. Den Momentanverbrauch berechne ich mit der PV-Anlage (Summe) und das geht sehr gut bis jetzt mit einem Python Script das den letzten empfangenen Wert verwendet, bis ein Timeout (30 sek.) den Wert auf 0 setzt (Wechselrichter defekt, offline, kein Licht etc). Das baue ich gerade auf NodeRed um, ist aber im Prinzip das gleiche.

mruettgers, 2022-11-17:
Ja genau, den Bezugswert meinte ich. Den lasse ich nämlich gerade per Telegraf und MQTT-Input-Plugin an InfluxDB weitergeben. Da würde ich die fehlenden Werte nämlich vermissen, wenn es keine Änderung gibt. Zumal ich da dann nicht so einfach mit Timeouts unterscheiden kann, ob es ein Problem gibt.

geronet1, 2022-12-10:
Hab mir das mal genauer angeschaut und versucht zu implementieren auf Basis von #54. Ist aber nicht trivial da es keine einfache Möglichkeit gibt innerhalb deines Codes die Werte von jedem Sensor und jedem Obis-Wert zwischenzuspeichern um ihn beim nächsten mal zu vergleichen ohne die libsml zu ändern. Im Prinzip hast du für Echtzeitanzeigen auch Recht daß das ständige senden eher besser ist, per NodeRED kann man ja einen Filter auf Wertänderung setzen um es dann in Influx zu speichern.

closed 2023-03-15 by mruettgers