mruettgers/SMLReader #46: No readout

kitt2011, 2022-05-18:
Hello again,

i tried the programm with a self compiled and you finish generatet Binary. Both of them dosnt have a readout to my mqtt server. The only message is a "hello start up message" to my mqtt server.
I have a smart meter from Landis&Gyr the E320.
Actually i use a Wemos with Tasmota and the SML driver an this works perfectly. The only reason why its so important for me to use youre programm is the live sending intervall. My tasmota solution only send all 10 seconds. I control my photovoltaic system (grid tie inverter) with my server but 10 seconds to late for zero feed.
I use the serial monitorbut there are also no messages.

mruettgers, 2022-05-18:
From what I could find out about the E320, there should be no problem running it with SMLReader. Did you use the circuit that you had in use with Tasmota? Or did you build it exactly as in my documentation? When SML data is recognized, the built-in LED of the Wemos D1 should blink several times. 

You could also try to activate the flags ` -DSERIAL_DEBUG=true -DSERIAL_DEBUG_VERBOSE=true` in the build target in ` platformio.ini` and then have another look at the output of the serial monitor.

kitt2011, 2022-05-19:
Exactly like your documentation. I have flashed your compiled binary. The LED is on input D2
The Internal LED from my Wemos light all the time after startup, no blinking

kitt2011, 2022-05-19:
now ive testet it only the wemos with your binary, no resistor and no IR LED.
The internal LED continusly lightning

Jabberwock-v6, 2022-11-05:
Hallo, 
I have exactly the same behavior on my miniD1 with just fototransistor and resistor. 
I get the hello messages from it but the led is on all the time. The meter is a eBZ DD3

kitt2011, 2022-11-11:
You need a IR lED no phototransistor

michaelosthege, 2022-12-25:
Hi @Jabberwock-v6, did you make any progress here? We confirmed the voltage changes with our PT333-3C on an eBZ DD3 too.

michaelosthege, 2022-12-27:
According to [the eBZ DD3 documentation section 6.1](https://www.manualslib.de/manual/409099/Ebz-Dd3.html?page=11#manual) we might need to use a different setting for the software serial: `SWSERIAL_7E1`.

However, with either setting we never get the expected start sequence.

By printing the read data with `Serial.printf("%02x", this->buffer[this->position]);` we can see the received bytes.
Even though there's a repetitive pattern, this is dependent on the resistor, and in no case we saw the start/end sequences.

michaelosthege, 2022-12-30:
We got things working: https://github.com/michaelosthege/obis_exporter

Learned a lot from the SMLReader project structure! We ended up writing something from scratch because the protocol is different, and we wanted to export the data as Prometheus metrics instead of MQTT messages.