mruettgers/SMLReader #52: main.cpp.o missing

Jack77777777, 2022-11-01:
When i try to build i get the below error and don´t know how to fix it because the file is not present.

[image]

Jack77777777, 2022-11-02:
WHen i look in src/main.cpp i see many "errors". I think this is something with include pathes but can´t find a solution. Hoepfully someone can help me.

[image]

Jack77777777, 2022-11-02:
No one any idea? Is this because i have nbo wemos attasched to the pc. I am waiting for my order to be delivered and wantedt to build in the meantime for the use with 3 heads.

mruettgers, 2022-11-02:
Hey @Jack77777777 ,

can you paste the contents of the platformio.ini?

Jack77777777, 2022-11-02:
@mruettgers thank you for the response.
platformio.ini:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[common]
platform = espressif8266@2.3.1
lib_deps =
    git+https://github.com/volkszaehler/libsml
    EspSoftwareSerial
    MicroDebug
    IotWebConf@^2.3.3
    ESPAsyncTCP
    git+https://github.com/philbowles/PangolinMQTT.git#v1.0.0
    jled

env_default = d1_mini
build_flags = -DIOTWEBCONF_PASSWORD_LEN=65
lib_ldf_mode = deep+

[env:d1_mini]
platform = ${common.platform}
board = d1_mini
framework = arduino
lib_deps = ${common.lib_deps}
lib_ldf_mode = ${common.lib_ldf_mode}
build_flags = ${common.build_flags} -DSERIAL_DEBUG=false
monitor_speed = 115200

[env:d1_mini_debug]
platform = ${common.platform}
board = d1_mini
framework = arduino
lib_deps = ${common.lib_deps}
lib_ldf_mode = ${common.lib_ldf_mode}
build_flags = ${common.build_flags} -DSERIAL_DEBUG=true -DSERIAL_DEBUG_VERBOSE=false
monitor_speed = 115200

[env:d1_mini_dev]
platform = ${common.platform}
board = d1_mini
framework = arduino
lib_deps = ${common.lib_deps}
lib_ldf_mode = ${common.lib_ldf_mode}
build_flags = ${common.build_flags} -DSERIAL_DEBUG=true -DSERIAL_DEBUG_VERBOSE=true
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
`

mruettgers, 2022-11-02:
Hm, looks ok. I'll try to do a fresh build later this evening.

Jack77777777, 2022-11-02:
@mruettgers All right. I am here for testing.

Danke

Jack77777777, 2022-11-02:
Is this the reason for my issue?
[image]

Jack77777777, 2022-11-02:
My screenshots above are from a fresh install on another pc. However, after clicking on platformio icon on the left menu it showed "building platfromio core" or something similar. After a while i could compile successfull. 

It shows me many problems but as it was compiled successfull i assume i can ignore this ?
[image]

mruettgers, 2022-11-02:
Most warnings are generated by the third-party library "PangolinMQTT". However, this should not be a problem.
Nevertheless, I replaced PangolinMQTT with another library in the develop branch and the branch needs some testers anyway. So feel free to test ;-).
The version in the develop branch also fixes an issue with reconnecting after WiFi loss.

Jack77777777, 2022-11-02:
Just comliped develop [d888efc](https://github.com/mruettgers/SMLReader/commit/d888efc507ed1e0ad3c7cf471fdae3caa8f8d341) with only 2 problems ;)
[image]

Jack77777777, 2022-11-02:
I think my issue was that i installed the system version of VS Code. With the User Version it is working on the other PC now as well.

closed 2022-11-02 by Jack77777777