mruettgers/SMLReader #56: How to activate internal pull-up resistor when using more than one sensor?

GreaseMonkey88, 2022-11-15:
Excuse my question, but I'm not too deep into programming. If I add another sensor, like in the circuit below, it will always pull both pins to ground as soon as one sensor opens. So my approach would be to activate the internal pull-up resistors to the sensor pin.

[image]

hamster65, 2022-11-15:
Your circuit is wrong. Do not connect D5 to D2. You have to use two external resistors between the Pins and 3,3V. Or activate the internal pull up resistors on both pins.

GreaseMonkey88, 2022-11-15:
Ok, I´ll try a different circuit.
But where in the code would I activate the internal pull up resistors on both pins?