mruettgers/SMLReader #57: No mqtt data

mscifm, 2022-11-25:
enviroment:
Nodemcu ESP8266
Smart Meter ISKRA MT691 INFO mode enabled

I could uploaf the FW succesfully. the IR transistor is conected to GPIO 4 i2C_SDA. 

The SMLReader connects to the MQTT Broker ( in this case IOBroker ), the Value LWT and info appears but no SMART Meter Voltage/Current information.

Could you assist me

mscifm, 2022-11-26:
After activating of the debugging I receive the following message
_Did not receive an SML message within 30 seconds, starting over.
State of sensor 1 is 'WAIT_FOR_START_SEQUENCE'.__

hamster65, 2022-11-26:
Maybe something is wrong with the positioning of the sensor or the circuit. Polarity, pullup, wiring....
My approach would be
- check if the meter is actually transmitting (use a phone camera, it should display purple flashes every second. Verify that the phone's camera can pick up infared with the remote control of your TV. Some cameras have an IR filter.
- check polarity, pullup value and wiring
- check the sensor's signal with an oszilloscope. Maybe something can be seen with a multimeter.