mruettgers/SMLReader #59: Decimal point is wrong in 1.8.0

AndyWo83, 2023-03-09:
The decimal point is wrong for data point 1.8.0. 
Instead of 4039.4 there should be 4.039. 
I have a new ISKRA eHZ MT681 that has the same SML parameters as the EMH ED300L you are using. 
Even with my previously used Hager EHZ363, the decimal point was also wrong. 
That's why I'm currently using Tasmota and there smartmeter integration, but I would like to switch to this, since Tasmota only sends the data via MQTT every 10 seconds.

[image]

emmrichd, 2023-04-24:
Hello,

Tasmota can send the data more often. you have to add 16 to the precision definition in the M-statement:
`<precision> | number of decimal places. Add 16 to transmit the data immediately. Otherwise it is transmitted on TelePeriod only.`
Regards, Dieter

btmerz, 2023-08-21:
Hi,
to get this, I added the lines
           if (scaler<0) {
              scaler -=3;   //convert values from Wh to kWhum
              prec=3;
            }  

at line 103 of MqttPublisher.h to convert the values from Wh to kWh.

Maybe that helps someone...