mruettgers/SMLReader #60: SSL/TLS support

sec-hack-research, 2023-03-31:
Hi, 

Thank you for your project, I've already successfully managed to transfer powermeter data to my MQTT server. For me the next step would be to enable some more security, and thus I am evaluating the possibilities of SMLReader with TLS. 

I've tried to find a way to activate TLS with SMLReader on my own. Would it be possible someone update the README on this topic with a small HowTo (ideally including a fingerprint verification), since I am still quite a beginner with the ESP8266? Also, I am seeing people, claiming that the ESPAsyncTCP has some bugs regarding SSL. Can someone verify this - and if so, adapt the SMLReader release to use an updated version of the lib? 

Thank you very much in advance! Great project you've provided here. Thank you!

emmrichd, 2023-04-01:
I strongly doubt, that an esp8266 has the necessary cpu performance to do this.

sec-hack-research, 2023-04-01:
> I strongly doubt, that an esp8266 has the necessary cpu performance to do this.

Both is clearly possible. Have a "service-application" on the ESP8266 that allows incoming TLS connections, and also it is possible to use the ESP8266 with outgoing requests to any server that supports TLS.